androidthings/sample-simplepio #2: Request documentation for Peripheral Manager objects from NDK

hpsaturn, 2016-12-27:
Hi, I was testing GPIO with last Google Things image via /sys/class/gpio but I must run it as root user. Whats is correct way for call "PeripheralManagerService" from C++ (NDK) and grant right permissions? In general, where is documentation for NDK layer?

Many thanks, great job guys!

hpsaturn, 2017-01-04:
I’m answering to myself, the last strategy used was: I used reflection so from the native layer I can access the Java layer and control the GPIO’s exposed by the framework (Peripherical Manager Service) but the performance is really bad.

My testing code (~300 kb firmware) takes around a minute in Raspbian (using the C++ driver and root user) but in Android Things it is taking around an hour (C++ - Java hybrid solution). Additionally, I tested with wiringpi and it only takes a few seconds (on Raspbian without sysfs gpio)

mangini, 2017-01-04:
@hpsaturn You are correct that the best way at the moment is going through the Java layer. We have plans to expose NDK direct bindings in the future. Can you please report a feature request in [b.android.com](https://code.google.com/p/android/issues/entry?template=Android%20Things%20feature%20request) so that you can get updates? Thanks!

closed 2017-01-04 by mangini

hpsaturn, 2017-01-04:
Many thanks for answer, the feature request is ready:
https://code.google.com/p/android/issues/detail?id=231484&thanks=231484&ts=1483569714