androidthings/sample-simplepio #8: Error on  openGpio(pinName) Caller lacks required permission com.google.android.things.permission.USE_PERIPHERAL_IO

ddave-22, 2018-05-13:
Facing Problem While running first blink app with RbPi 3 
` Caused by: java.lang.SecurityException: Caller lacks required permission com.google.android.things.permission.USE_PERIPHERAL_IO
                                                                                       at android.os.Parcel.readException(Parcel.java:2004)
                                                                                       at android.os.Parcel.readException(Parcel.java:1950)
                                                                                       at com.google.android.things.pio.IPeripheralManagerClient$Stub$Proxy.GetGpioClient(IPeripheralManagerClient.java:759)
                                                                                       at com.google.android.things.pio.GpioImpl.<init>(GpioImpl.java:50)
                                                                                       at com.google.android.things.pio.PeripheralManager.openGpio(PeripheralManager.java:197)
                                                                                       at com.example.androidthings.simplepio.BlinkActivity.onCreate(BlinkActivity.java:51)`

JesseScott, 2018-05-14:
I am also seeing this - and also for the code at https://developer.android.com/things/training/first-device/peripherals#blink_an_led

I have added the permission as a parent of the main <manifest> tag and have restarted the device (as i believe that Things is supposed to grant permssions on startup ?)

I have toggled Instant Run off and On too, as some Stack Overflow posts have suggested (for a different permission error) to no effect.

Fleker, 2018-05-16:
Which version of Android Things are being run on your device?

ddave-22, 2018-05-18:
@Fleker  Sorry for delayed response Android Things 1.0 it is.

JesseScott, 2018-05-20:
Haha. It was a typo for me - I had USE_PERIPHERAL_ID instead of _IO

ddave-22, 2018-06-01:
Alright i guess it Should work after manually permission. thanks for your support

closed 2018-06-01 by ddave-22

dinusha94, 2019-06-10:
Use this Permission settings in Manifest.xml
<uses-permission android:name="com.google.android.things.permission.USE_PERIPHERAL_IO"/>