androidthings/sample-simplepio #9: how to control more than 2 servos?

Mao-x-w, 2018-06-15:
how to control more than 2 servos?because there were only two pwm(pwm0 and pwm1)

Mao-x-w, 2018-06-19:
Can we use the GPIO port to simulate PWM output in Android Things.Like this:
`void servo_pulse(int myangle)
{
  int PulseWidth;                    //定义脉宽变量
  PulseWidth = (myangle * 11) + 500; //将角度转化为500-2480 的脉宽值
  digitalWrite(ServoPin, HIGH);      //将舵机接口电平置高
  delayMicroseconds(PulseWidth);     //延时脉宽值的微秒数
  digitalWrite(ServoPin, LOW);       //将舵机接口电平置低
  delay(20 - PulseWidth / 1000);     //延时周期内剩余时间
  return;
}`
I can't make the thread to sleep 0.5ms in Android Things.

Fleker, 2018-06-19:
The GPIO control in Android Things doesn't have the level of precision that you may want for high frequency PWM.